urlgrey/hsmm-pi #39: git clone routine not able to find bbhn packages

John-Richards, 2015-01-29:
When installing  through the install.sh routine an error occurs when it attempts to run "git clone git://ubnt.hsmm-mesh.org/bbhn_packages". The error is a failure to connect to the url... I checked, it no longer exists. It doesn't appear that they use any of the same directory structures at broadband-hamnet.org anymore. Any suggestions?
John, KY8T

John-Richards, 2015-02-04:
Problem solved. BEFORE!!! running install. sh you MUST edit the line calling the bbhm pckages from:
git clone git://ubnt.hsmm-mesh.org/bbhn_packages
To:
git clone git://ubnt.broadband-hamnet.org/bbhn_packages

argilo, 2015-03-03:
Recently removed the dependency on bbhn_packages completely, as the patch I was pulling from that repo is no longer needed. So this shouldn't be an issue anymore.

closed 2015-03-03 by argilo