urlgrey/hsmm-pi #47: [BUG] When bppting.

deavmi, 2015-04-26:
After installing HSMM-Pi on Raspbian on my RapsberryPi's, I then reboot, ut loose all connectivity to my Pi.

clayton-shopify, 2015-04-26:
Try connecting to 172.27.2.1, which will be the Pi's IP address after it reboots. Remember that the Pi will be acting as a DHCP server, so you should connect your computer straight to it and get an IP address from it. You should also be able to connect to it by name: localnode.

argilo, 2015-04-26:
Try connecting to 172.27.2.1, which will be the Pi's IP address after it reboots. Remember that the Pi will be acting as a DHCP server, so you should connect your computer straight to it and get an IP address from it. You should also be able to connect to it by name: localnode.

deavmi, 2015-04-26:
It was on my LAN before the reboot, so should I now connect to it via direct ethernet (from my laptop plugged into Pi via ethernet)? :smile:

argilo, 2015-04-26:
Yes, you'll have to take it off your LAN and connect directly.

deavmi, 2015-04-26:
@argilo Okay, thanks :smile:

deavmi, 2015-04-26:
@argilo Still no success... :cry:

argilo, 2015-04-26:
Do you have all other network interfaces on your laptop (e.g. WiFi) turned off? What IP address does your connection to the Raspberry Pi have?

deavmi, 2015-04-27:
I turned off my WiFi on my laptop and connected directly from  y laptop's ethernet port to my RaspberryPi's thernet port. Still no success, I tried visiting the page you specefied.

argilo, 2015-04-27:
Check whether your laptop was assigned an IP address by the Raspberry Pi. It should begin with 172.27.2. If that looks OK, make sure you're using port 8080 in the URL when you try to access hsmm-pi.

deavmi, 2015-04-27:
@argilo Oh, I think it was my old ethernet settings that were messing things up! :small:

deavmi, 2015-04-27:
@argilo I have re-initialised them and the Default Route label now shows 172.27.2.1

argilo, 2015-04-27:
Excellent. See if you can access http://localnode:8080/ or http://172.27.2.1:8080/.  You should also be able to log in through SSH if needed.

argilo, 2015-04-27:
Glad to hear it's all working now. If there are things that need to be clarified in the README, let me know.

deavmi, 2015-04-27:
@argilo Going to test the mesh now, set the WiFi Broadcast Channel to 9, got the same BSSID, so I think things shall be going well after this reboot. **Thanks!** :smile:

deavmi, 2015-04-27:
@argilo Maybe an FAQ section could be good, on the repo's wiki. :smile:

deavmi, 2015-04-27:
@argilo They do not seem to be picking each other up for some reason though. There is an ad-hoc  BSSID being braodcast, which I set to (on both the nodes) _"WMeshNet"_, but no links are shown on the status page.

argilo, 2015-04-27:
Are you using hsmm-pi on both ends?

deavmi, 2015-04-27:
@argilo Yep :smile:

deavmi, 2015-04-27:
@argilo Something like this would be a good FAQ, https://github.com/urlgrey/hsmm-pi/wiki/FAQ

argilo, 2015-04-27:
Double check that your settings are identical on both sides, apart from the node name which should be different. Also you can check the olsrd status at http://local node:1978/all for signs of trouble.

deavmi, 2015-04-27:
@argilo There seems to be nothing under the "_neighbours_" section.

argilo, 2015-04-27:
Does the "Interfaces" section look okay on both ends?

deavmi, 2015-04-27:
I just generated a config file of Olsrd's current config, and everything seems to be marked out a.k.a `#`, surely this must be the problemo:

````
#
# Configuration file for olsr.org - pre-0.6.8.1-git_9891303-hash_7da0af84d0db0f920462d6ccb4f41102
# automatically generated by olsrd-cnf parser v. 0.1.2
#

# OLSR.org routing daemon config file
# This file contains ALL available options and explanations about them
#
# Lines starting with a # are discarded
#

#### ATTENTION for IPv6 users ####
# Because of limitations in the parser IPv6 addresses must NOT
# begin with a ":", so please add a "0" as a prefix.

###########################
### Basic configuration ###
###########################
# keep this settings at the beginning of your first configuration file

# Debug level (0-9)
# If set to 0 the daemon runs in the background, unless "NoFork" is set to true
# (default is 1)

DebugLevel  0

# IP version to use (4 or 6)
# (default is 4)

# IpVersion 4

#################################
### OLSRd agent configuration ###
#################################
# this parameters control the settings of the routing agent which are not
# related to the OLSR protocol and it's extensions

# Clear the screen each time the internal state changes
# (default is yes)

# ClearScreen     yes

# Should olsrd keep on running even if there are
# no interfaces available? This is a good idea
# for a PCMCIA/USB hotswap environment.
# (default is yes)

# AllowNoInt  yes

# LockFile
# The lockfile is used to prevent multiple OLSR instances running at the same
# time.
# (Linux/BSD default is "/var/run/olsrd-ipv(4/6).lock")
# (Win32     default is "configfile-ipv(4/6).lock")

# LockFile "lockfile"

# Polling rate for OLSR sockets in seconds (float). 
# (default is 0.05)

# Pollrate  0.05

# Interval to poll network interfaces for configuration changes (in seconds).
# Linux systems can detect interface statechange via netlink sockets.
# (default is 2.5)

# NicChgsPollInt  2.5
[... 497 lines cut ...]
    # but here you can specify a fixed value.
    # Olsrd will choose links with the lowest value.
    # Note:
    # Interface weight is used only when LinkQualityLevel is set to 0.
    # For any other value of LinkQualityLevel, the interface ETX
    # value is used instead.
    # Weight 0
    
    # If a certain route should be preferred
    # or ignored by the mesh, the Link Quality
    # value of a node can be multiplied with a factor
    # entered here. In the example the route
    # using 192.168.0.1 would rather be ignored.
    # A multiplier of 0.5 will result in a small
    # (bad) LinkQuality value and a high (bad)
    # ETX value.
    # Note:
    # Link quality multiplier is used only when
    # LinkQualityLevel is greater than 0.
    
    # LinkQualityMult 192.168.0.1 0.5
}

# END AUTOGENERATED CONFIG
````

argilo, 2015-04-27:
Not everything is commented out there. It looks normal to me.

deavmi, 2015-04-27:
@argilo Everything looks correct, on the configuration side, I am doing a manual reboot now.

deavmi, 2015-04-27:
@argilo Okay, that's good to know :smiley:

deavmi, 2015-04-27:
The USB Wifi adpater sems to be off on the one Pi.

deavmi, 2015-04-27:
Okay, both dongles are powered on, but still, there seems to be a problem.

argilo, 2015-04-27:
I've had problems with certain WiFi adapters not working so well. Maybe a power supply that's not supplying enough current?

argilo, 2015-04-27:
If you can't get it going, I'd suggest swapping in a different model of WiFi adapter to see if it works any better.

deavmi, 2015-04-27:
I think it is possibly antenna orientation. and such.

deavmi, 2015-04-27:
@argilo Might want to check this out too https://github.com/urlgrey/hsmm-pi/issues/50

deavmi, 2015-04-27:
@argilo This is what I get on the **Status** page. `Warning!. There are no mesh links in range. It's a bit quiet around here. `.

deavmi, 2015-04-27:
@argilo Was watching [this](https://www.youtube.com/watch?v=ltUAw02vfqk), and @urlgrey says that nodes annouce their prescenece every 10 minutes, so maybe I ought to be patient.

deavmi, 2015-04-27:
@argilo How many wifi adapters/antennas does pone node require?

deavmi, 2015-04-27:
@argilo 

> I've had problems with certain WiFi adapters not working so well.

Did they atleast mesh though?

argilo, 2015-04-28:
Nodes should mesh within a minute or two if things are working properly. Only one adapter/antenna per node is required. The problematic adapters I've seen work for a little while, then stop.

arnoldsmyth, 2015-05-20:
Did anyone ever come to a resolution on this? I have this exact same problem. The nodes can see the Mesh when I run "iwlist wlan0 scan".

I can connect to the internal Mesh node with the Ethernet cable and the 172.27.2.1 IP address but while it see's the Mesh with a "Wifi Scan" the Status page still has the "It's a bit quiet" message.
I can also connect to the gateway node with the IP address but it has the same issue.

The internal node does not have any access to the internet to run updates etc.

deavmi, 2015-05-20:
@arnoldsmyth I feel ya bro. Lol :smile:

mathisono, 2015-05-20:
> @arnoldsmyth <https://github.com/arnoldsmyth> I feel ya bro. Lol [image:

Do a cold reboot remove the power. Ive had this problem a few times,
sitting at my desk stumped my self, pull the power plug wait 30sec and
reapply.

73 mathison

arnoldsmyth, 2015-05-20:
@mathisono can you expand on what you mean by "reapply"?

mathisono, 2015-05-21:
plug it back in

argilo, 2016-04-23:
I'm closing this, as it appears the original issue was resolved.

closed 2016-04-23 by argilo