urlgrey/hsmm-pi #50: Setting of WiFi channel

deavmi, 2015-04-27:
Once the WiFi channel is set, it does not seem to actually get saved, as once I did a WiFi scan, the BSSID was pciked up running on channel `1` while I had set it to channel `9` and saved the setting.

deavmi, 2015-04-27:
**UPDATE:** The channel setting does nothing, checked on my phone, I know set the BSSID broadcast channel too `5`, but it shows up on channel `1`.

virtualdxs, 2015-05-30:
I just tested this. It successfully writes to /etc/network/interfaces, but whatever TXes doesn't look there apparently.

virtualdxs, 2015-05-30:
I'm trying to fix this but I can't find for the life of me what controls TXing. It would help if I could find out where `get_mesh_info()` was defined.

deavmi, 2015-05-30:
So it is a problem with a JavaScript method in the code? Possibly.

virtualdxs, 2015-05-30:
No. The JS is fine. Even the PHP that writes to the file is fine. The problem is the program that *transmits* doesn't read /etc/network/interfaces.

deavmi, 2015-05-30:
Oh. So it has a hard coded value of the transmitting frequency.

argilo, 2016-04-23:
I tested this with the Raspberry Pi 3's built-in WiFi, a Ralink 5370 adapter and an Alfa AWSU036NH. All of them changed channels as expected after a reboot.

If you're still able to reproduce this with the latest Raspbian and HSMM-Pi then please let me know what WiFi adapter you're using.

closed 2016-04-23 by argilo