urlgrey/hsmm-pi #51: Support for Docker

vielmetti, 2015-05-28:
The goal of this effort would be to take the build script and turn it into a Dockerfile, thus encapsulating all of the dependencies in such a way that someone could just take an image and load it up on their Pi.

I'll track my efforts here.

urlgrey, 2015-06-13:
This is a great suggestion, one that I've considered myself.  However, I don't think Docker is really well-suited for running the multiple services (apache, dhcpd, gpsd, etc) that are installed and meant to run with HSMM-Pi.

A lot of the issues reported with HSMM-Pi are related to people running disparate versions of Raspbian and packages installed by HSMM-Pi.  My plan is to create a Raspbian-based distribution of HSMM-Pi that includes all the required packages.  This will hopefully simplify the installation and eliminate the requirement that you have Internet connectivity to create a new HSMM-Pi instance.

lavvy, 2015-11-21:
Please any info for the distro?