urlgrey/hsmm-pi #52: Broadband Hamnet version 3

virtualdxs, 2015-05-29:
Broadband Hamnet version 3 was released for all hardware, but it's not compatible with v2 or v1. Would it be possible to get HSMM-Pi to work with BBHN3?

virtualdxs, 2015-05-29:
Label broadband-hamnet-compatibility should be assigned but I don't know how

virtualdxs, 2015-05-29:
I'd be very glad to help if I knew how, but I have no idea where to start.

argilo, 2015-05-29:
I haven't had any trouble using hsmm-pi with BBHN 3. Have you run into problems?

virtualdxs, 2015-05-29:
No, but I assumed that it was incompatible seeing as it hasn't been worked on in a month and the site says 3.x is incompatible with 2.x, which is why I haven't upgraded my Linksys nodes. So you're saying you have used it with 3.x and have not had problems?

argilo, 2015-05-29:
No trouble at all. There are several hsmm-pi nodes running on the local BBHN v3 mesh.

virtualdxs, 2015-05-29:
Alrighty then. I'm really surprised, but if it works, it works. Thanks! Closing the issue.

closed 2015-05-29 by virtualdxs