urlgrey/hsmm-pi #58: Problems installing on Raspbian-Jessie

SDreher71, 2015-12-04:
After install, localhost:8080 returns the Apache verification page. After editing /etc/apache/ports.conf to point the 8080 virtual host to /var/www, it returns a page with several of the hsmm-pi menu items, but with no css formatting. Clicking on any of the menu items on that page (status, login, etc) results in a 404 error.

argilo, 2015-12-07:
Thanks for reporting this. I've added support for Jessie in https://github.com/urlgrey/hsmm-pi/pull/60.

The issue was that Jessie configures Apache a bit differently. Configuration files are stored in `/etc/apache2/conf-available` instead of `/etc/apache2/conf.d`, and web content is stored in `/var/www/html` instead of `/var/www`.

hsmm-pi should still work fine on Wheezy as well.

closed 2015-12-07 by argilo

SDreher71, 2015-12-07:
Nice work. Thanks!

argilo, 2015-12-07:
By the way, you can use the new "Lite" version of Raspbian Jessie which omits lots of packages that hsmm-pi doesn't need.