urlgrey/hsmm-pi #61: Status not showing all Linksys routers in network

ghinton77, 2015-12-07:
I have 3 Linksys routers in my home network.  One is downstairs and the other 2 are in the same room as the HSMM-Pi node.  The HSMM-Pi only shows the 2 in the same room on its Status page.
Does HSMM-Pi only show the nodes it is actually connected to or all the nodes in the network?
One of the Linksys nodes is a tunnel.  On the other Linksys nodes I can see the Remote nodes on the other side of the tunnel.  On the HSMM-Pi Status page it only shows the tunnel.
So maybe that is two questions, but they are related.
Gary

argilo, 2016-04-02:
The HSMM-Pi status page only shows neighbours; remote nodes are not shown. I'll see about adding remote nodes soon.

argilo, 2016-04-04:
Version 0.8.0 now displays remote nodes. Are you able to see all your nodes on the status page now?

ghinton77, 2016-04-07:
Thanks for updating the program.

I am real busy on another project right now.  I will try to check it in a few weeks.

Gary

argilo, 2016-04-13:
Ok, I'll close the issue for now. Feel free to re-open it if the issue isn't fixed.

closed 2016-04-13 by argilo