urlgrey/hsmm-pi #64: Mesh network not being recognized on node

va3rjs, 2016-02-26:
I created 2 nodes on 2 identical PI's, one in gateway mode and the other in internal mode.

When I look at available wifi networks I see the opposite node being advertised at 100% quality.

The OLSRD is running but the status page shows there are no mesh networks that are in range.

I run the daemon from the command line with various settings of debug, but I don't get a clue as to what is wrong.

Anyone have any ideas?

Thanks
Bob
Va3rjs

trrhodes, 2016-03-22:
Having the same issue

va3rjs, 2016-03-23:
I ordered some new wifi dongles to see if that solves,the problem........I suspect that the ones I have work for wifi on the Pi but there may be some issues with the associated drivers.........

I found iwconfig commands do not seem to work on the dongles I have........

I hope the new ones get here this week....I will update you when I they arrive.

Bob

trrhodes, 2016-03-23:
Interesting, I have also ordered new dongles and am awaiting their arrival. Thanks for getting back to me. Look forward to hearing if it solves your issue.

Tyler

va3rjs, 2016-03-24:
Just received my new USB/WIFI dongles and now I can see other nodes in my MESH here at home...this is significant progress.......I will try setting up some services and see what happens.

Bob

trrhodes, 2016-03-24:
Thats great news. Which dongles did you get?

Tyler

va3rjs, 2016-03-24:
I used one with the  Ralink 5370 chipset......
looks the same as the one in this picture:
HSMM-Pi Blog

argilo, 2016-04-23:
I'm closing this ticket, as it seems to have been an issue with hardware compatibility.

closed 2016-04-23 by argilo

va3rjs, 2016-04-23:
Yes thanks it as the usb wifi dongle.

B