urlgrey/hsmm-pi #67: Service announcements

mathisono, 2016-03-25:
SO I can only define ports Ie http://KJ6DZB-4.local.mesh:8080

I cant announce http://KJ6DZB-4.local.mesh:8080/meshchat 

Ideas ?

mathisono, 2016-03-25:
the services are not showing up in the rest of the aredn mesh

argilo, 2016-03-25:
I haven't used that feature before. Just tried it and got the same result as you. I'll dig into it soon.

mathisono, 2016-03-31:
any update... I will check out the new update but. I want to sort this out... I will didg around in the olsrd config for Service announcements. The HSMM-pi web page shows the Service announcements but its not going out to the mesh. ;-(

argilo, 2016-03-31:
I haven't fixed this yet. I'll update this ticket if I learn anything more.

argilo, 2016-03-31:
OK, I think I tracked down the problem to a missing trailing slash on the service URL. Once I added that in olsrd.conf the service appeared in AREDN.

argilo, 2016-04-01:
@mathisono I just pushed a fix that should allow the rest of the AREDN mesh to see the hsmm-pi node's services. Next I'll have to add a path parameter to hsmm-pi.

argilo, 2016-04-02:
This should be fixed in version 0.7.0.

closed 2016-04-02 by argilo

mathisono, 2016-04-04:
nice work on the bug fix effort! I oh you a beer! I will give 0.8.0 a run