urlgrey/hsmm-pi #73: Network service does not advertise correct port.

dandrzejewski, 2016-04-02:
When I add a network service, like so:
[image]

And save it, it adds it like this:

[image]

Further, I don't see any port forwarding getting added in iptables.

Am I missing something or is this a bug?

argilo, 2016-04-02:
That looks correct. The forwarding rule should be written to /etc/rc.local. Is it not working after a reboot?

argilo, 2016-04-02:
"Port" is the port that is advertised, and "Local Port" is the port that the connection gets forwarded to behind NAT.

dandrzejewski, 2016-04-02:
Ah. Got it. I had them backwards.  And, also, I was neglecting the "-t nat" when I was listing my iptables output.

I think the labelling on the UI for the ports should be more clear.  Aside from that, all is working well.

closed 2016-04-02 by dandrzejewski

argilo, 2016-04-02:
Any suggestions for the labelling? I just changed the labels because I thought they were unclear before ("forwarding port" for the externally visible port, and "port" for the one behind NAT). I chose the new labels to match the terms my wifi router uses for the same.

argilo, 2016-04-02:
BTW I'm planning on implementing the 5-host direct & 13-host direct modes soon, which will make NAT unnecessary.