urlgrey/hsmm-pi #76: How to Add a Service

Vhfcomm, 2016-04-03:
Hello,

I am trying to add a service, but don't know which settings to use. On the Add Network Service page, it asks for Service Protocol Name and Port but I don't know what these should be. I am trying to set up Mumble (open source keyboard and chat service) and can access it locally , i.e. on the machine with the Mumble server on. However I cannot access it from the remote machine. What should Service Protocol Name be and the Port, which is presumably the port on the R_Pi which should be open? Other settings appear to be correct.

Thanks

Graham

argilo, 2016-04-03:
Service protocol name is whatever protocol your service is accessed by, probably `http`. Port is the port number that the Raspberry Pi will forward to your service. Normally that can be set to the same port your service is listening on. The port your service is listening on should go in the "local port" field.

Hope that helps.

argilo, 2016-04-03:
By the way, the "service protocol name" only affects the URL that is announced on the mesh.

ehresman, 2016-04-03:
Mumble protocol defaults to port #64738.

Vhfcomm, 2016-04-03:
Thank you for your help, I will try those settings tomorrow when I have more time. AIUI, I set both ports to 64738 and service protocol to http. How would I know which protocol is used to access Mumble? I've Googled for information, but can't find anything other than TCP and UDP.

ehresman, 2016-04-03:
The protocol is called mumble. It is definitely not a varient of http, more
like irc with significant extensions to support VoIP and secure
authentication.

73;
Bob KV4PC

Vhfcomm, 2016-04-03:
Thank you Bob. I had tried mumble as the protocol name and had made some progress with that, but I was just fumbling in the dark and with no documentation, I was trying anything. Will try again tomorrow.

73
Graham
GW8RAK

Vhfcomm, 2016-04-04:
Evening all. Still cannot connect to the Mumble server from the remote machine. The error is:

" Server connection failed: Connection refused."
This means there is a computer there but that is the incorrect port, double check the IP to make sure that this is the right computer, if it is then check what port you are supposed to connect on and put that in the port box on the connect screen.

Should both the Port and Forwarding Port be set to 64738? I can connect from the local machine (the one with the Mumble server on), but not from the remote machine.

Graham

ehresman, 2016-04-04:
If you are crossing a NAT style firewall then you will need some form of
port forwarding on the firewall for most protocols that you intend to use
across the firewall. The direction you cross the firewall from e.g. inside
to out vs outside to in may affect the type of forwarding that you choose.

73;
Bob KV4PC.

Vhfcomm, 2016-04-04:
Thanks Bob.You say "type of forwarding". Are there different types or do you just mean different ports? 

Graham

argilo, 2016-04-04:
Try these settings:
* Service Protocol Name: mumble
* Host: IP address that the Raspberry Pi assigns to the machine running your mumble server; it should be something like 172.27.2.5
* Port: 64738
* Protocol: TCP
* Local Port: 64738

Also create add a second entry but use UDP in place of TCP.

You may also need to set a path of `?version=1.2.0` but it seems a default version will be assumed if you don't fill that in.

When accessing the mumble server, be sure to use the Raspberry Pi's node name or IP address (10.x.x.x).

argilo, 2016-04-04:
And of course, be sure to reboot the Raspberry Pi after applying the settings above.

ehresman, 2016-04-05:
Firewalls usually have mapping ports and trigger ports. There are other
types as well.

Vhfcomm, 2016-04-05:
Argilo,
I've tried those settings and rebooted, but still cannot connect. I will try the set path option this evening. If that fails, I will try to just write browser to browser to keep it the simplest possible.
Thanks for your help guys.

Vhfcomm, 2016-04-18:
Been away a lot recently, so not had much time to continue with HSMM-Pi. I cannot get Mumble to work at all with any permutations of settings, so it could be that Mumble just doesn't like HSMM-Pi. While I try to get my desired service going, does anyone have a working service and settings which I can try please? If you can let me know what the service is and how it is configured, it will help to prove that HSMM-Pi is working.