urlgrey/hsmm-pi #78: Get rid of callsign announcement?

argilo, 2016-04-04:
@urlgrey Broadband Hamnet and AREDN don't do separate callsign announcements the way hsmm-pi does, presumably because the convention is to put one's callsign into the node name which is announced very frequently anyway. Do you think it make sense to simplify hsmm-pi by removing the callsign announcement feature?

urlgrey, 2016-04-04:
That sounds good to me. Including the callsign in an ICMP ping packet
always seemed like a hack; the convention of embedding the call sign in the
node name is preferable.

argilo, 2016-04-04:
OK, I've removed it in my fork and will put that change in the next release: https://github.com/argilo/hsmm-pi/commit/134bbc5777faf988f9d894afd19e90f8985982ae

closed 2016-04-26 by argilo