urlgrey/hsmm-pi #79: Support "direct" LAN mode in addition to NAT

argilo, 2016-04-04:
NAT is a pain, so I'd like to implement "direct" LAN mode like in Broadband Hamnet & AREDN. In this mode, each host connected to the LAN gets its own IP address on the mesh, and no port forwarding is required.

cbegg50, 2016-05-29:
I have opened a fork and a pull request and have Direct mode working, including a new reservation screen, similar to BBHN's Linksys routers.  Until its accepted by urlgrey, you can get it at https://github.com/cbegg50/hsmm-pi.git.

pumatrax, 2017-02-14:
I tried your fork cbegg and the script dosent get it installed. Would of loved to check out direct mode

pumatrax, 2017-02-18:
Took a lot of figuring out because I am new to hsmm-pi and even Linux but I got it working. I like your additions.