urlgrey/hsmm-pi #80: Error in new version

shaozhai, 2016-04-19:
[image]

argilo, 2016-04-20:
@shaozhai Can you provide some further details about when this error appeared? Were you trying to install on top of a fresh Raspbian Jessie install on a Raspberry Pi, or something else?

argilo, 2016-04-24:
Also, if you could paste the entire output of the install process, that would be very helpful.

shaozhai, 2016-04-25:
raspberry pi2 debian

argilo, 2016-04-25:
Was it a fresh install of Raspbian?

Which version of Raspbian were you using? Wheezy or Jessie? What was the release date?

Please paste the full output of the install process.

Without the above information, I can't help.

shaozhai, 2016-04-27:
Jessie

argilo, 2016-04-27:
Sorry, you haven't provided enough information to diagnose the issue so there's not much I can do.

closed 2016-04-27 by argilo

shaozhai, 2016-04-27:
The system did not find this  database file ‘hsmm-pi.sqlite’

shaozhai, 2016-04-27:
[image]

shaozhai, 2016-04-27:
The new version of the re installation problem is still

shaozhai, 2016-04-27:
Did not create this file hsmm-pi.sqlite

argilo, 2016-04-27:
@shaozhai You must start with a fresh install of Debian Jessie.

argilo, 2016-04-27:
Erase your SD card and start from scratch.

shaozhai, 2016-04-27:
[image]

shaozhai, 2016-04-27:
Have you installed the new HSMM-PI yourself? 0.6 without this problem

argilo, 2016-04-27:
Yes, many times.

shaozhai, 2016-04-27:
0.8 there is a problem.

shaozhai, 2016-04-27:
Can I  manually create this hsmm-pi.sqlite?

argilo, 2016-04-27:
Erase your SD card, install 2016-03-18-raspbian-jessie-lite, install HSMM-Pi and everything should work fine.

shaozhai, 2016-04-27:
I try!