urlgrey/hsmm-pi #81: Lite distribution has no X so can't do UI!

cbegg50, 2016-04-20:
Your suggestion of Raspbian Jesse Lite does not have the X that the Admin interface uses.  I've had to refresh the SD card with Standard Jesse to use the script.  I'd already used a NOOBS installation on a Pi Zero, but for the second Pi node I thought I'd follow your instructions.  Bad idea1
Thanks for an otherwise superb product.  My brother G7AOR is also trying one and he is having problems with DHCP crashing the image.

73 de Colin, G4MCF

argilo, 2016-04-20:
@cbegg50 Did you use the full Raspbian Jessie for your first install? Did everything work well? If so, I can add that as an option to the instructions for people who want to use the GUI.

mathisono, 2016-04-20:
I don't use the light image unless I specifically need a head less node. I
run an aprs server so I need the ui.

73 kj6dzb

Reopened #81 <https://github.com/urlgrey/hsmm-pi/issues/81>.

closed 2016-04-23 by argilo