urlgrey/hsmm-pi #88: Remove gpsd-clients?

argilo, 2016-04-26:
@urlgrey The install script installs the `gpsd-clients` package, which causes a lot of dependencies to be installed:

https://github.com/urlgrey/hsmm-pi/blob/00ad788cc7b2f9f16bdccb4c801dc807c1ac1db4/install.sh#L35

Is that package needed? Here's what it includes, and I don't see any of that used in HSMM-Pi itself:

https://packages.debian.org/jessie/armhf/gpsd-clients/filelist

Since you're using the GPS functionality, maybe you included it for another reason?

urlgrey, 2016-04-26:
I've found the gpsmon utility useful for troubleshooting GPS receivers.
It's not critical to the operation of HSMM-Pi though. I'm okay with
removing the gpsd-clients package from the standard installation.

mathisono, 2016-04-26:
Its better to have gpsd than to not. 

I run xastir on my hsmm-pi node and use gpsd to host the usb gps puck, allowing multiple programs to get gps data, and even distribute it too local lan. 

gpsd is also used with ntpd to set a stable time reference. Its a standard package on most of my rpi's.    

On the other hand the AREDN distro has chosen out of simplicity to just set a static lat long, then its reported in a jason message for a mapping script. This removes the need for a olsrd plugin to report the location. As that a rpi might move (but not always), I have a portable aprs server/digipetor thats not static. If i move I dont have to go update the lat/long. Problem I see is that the olsrd location plugin isn't really needed. The code could adopt what ARDEN is doing with Jason messages and use gpsd as an option to manage a gps device or just let the user set it static.

argilo, 2016-04-26:
@mathisono I wasn't proposing to remove gpsd, just gpsd-clients (which HSMM-Pi doesn't itself doesn't use). My motivation for doing so is that it causes dozens of extra packages to be installed, thus slowing down the install process.

closed 2016-04-29 by argilo