urlgrey/hsmm-pi #91: Direct Mode, OLSR Status Page, dhcp reservations as OLSR Hosts

cbegg50, 2016-05-10:
Hi, my brother Rodney and I have been working on modifying hmm-pi to support Direct as well as NAT mode on the LAN, adding a link to the OLSR Status Page and adding DHCP reservations as OLSR hosts, all of which were missing from the published version.  I've completed the work and would now like to commit the changes.  To this end, I've created a fork at cbegg50/hsmm-pi and will put my changes in there.  Changed files are:
/webroot/files/olsrd/olsrd.conf.templage
/Model/NetworkSetting.php
/View/Layouts/default.ctp
/View/NetworkSettings/edit.ctp
/Controller/AppController.php
/Controller/NetworkSettingsController.php

I'm new to web based Git, where do I go from here?

Regards,
Colin Begg

Tom-McDermott, 2016-05-12:
Hi Colin - not involved in hsmm-pi (well, not yet).
With github, in order to contribute your code changes:

1. Fork hsmm-pi to your own github page  (you've apparently done that).
2. Clone from your github to your local machine.
3. Make and test your changes locally.
4. Commit your local changes.
5. Push your local branch to your github.
6. Go to the hsmm-pi github and do a pull request.  It will know you forked a copy, and
thus to pull from your github.
7. Wait for the maintainer of hsmm-pi to review and if agreeable, to Merge your changes.
8. The maintainer may or may not agree, of course.

-- Tom, N5EG

cbegg50, 2016-05-12:
Thanks, Tom.  In the meantime, I've done the first few bits but named the directory hmm-pi-direct.  Should I keep the original hmm-pi directory name?

Colin, G4MCF

Tom-McDermott, 2016-05-12:
I think github (and git) require the original directory naming be exactly the same. That's one reason for forking the original.  Many folks will git checkout a new branch to put their changes on, and push that branch up to their github and thence pull request to origin. Then the maintainer will be able to decide whether to keep a separate branch, or to merge into the master branch.

-- Tom, N5EG

cbegg50, 2016-05-14:
I've started working on my fork now, so naming should be fine.  I should be able to pull down my changed files one by one and add them to branches on my fork.

Thanks, Colin, G4MCF

virtualdxs, 2016-05-19:
The directory name does NOT have to be even similar. Everything is in the .git folder. If you have this problem in the future, I recommend:

1. Git clone the repo
2. Copy your changes over to the new repo folder
3. Git commit (make sure everything is as you want it first)

The name of the directory doesn't matter at all.

@urlgrey You may want to close this issue.

closed 2016-05-25 by cbegg50