urlgrey/hsmm-pi #95: Internal Error occured

sv1, 2016-05-28:
Hello I am getting a Internal Error Occurred when trying to Login, after upgrading php5 to version 5.6.20 through apt-get update on a Debian Jessie.

Any suggestions would be greatly appreciated.

sv1, 2016-05-28:
This is the error when trying login manually
$ php -f login.ctp 
<div class="users form">
PHP Fatal error:  Using $this when not in object context in /home/pi/hsmm-pi/src/var/www/hsmm-pi/View/Users/login.ctp on line 3

cbegg50, 2016-05-29:
For internal errors, look in ~/hsrmm-pi/src/var/www/hsmm-pi/logs at error.log, there's a traceback in there.

sv1, 2016-05-29:
Although shockwave flash plugin is installed on Iceweasel, I get the error below.

2016-05-29 20:47:07 Error: [MissingComponentException] Component class FlashComponent could not be found.
Exception Attributes: array (
  'class' => 'FlashComponent',
  'plugin' => false,
)
Request URL: /hsmm-pi/users/login
Stack Trace:
#0 /usr/share/php/Cake/Controller/ComponentCollection.php(53): ComponentCollection->load('Flash', Array)
#1 /usr/share/php/Cake/Controller/Controller.php(643): ComponentCollection->init(Object(UsersController))
#2 /usr/share/php/Cake/Routing/Dispatcher.php(188): Controller->constructClasses()
#3 /usr/share/php/Cake/Routing/Dispatcher.php(167): Dispatcher->_invoke(Object(UsersController), Object(CakeRequest))
#4 /home/pi/hsmm-pi/src/var/www/hsmm-pi/webroot/index.php(110): Dispatcher->dispatch(Object(CakeRequest), Object(CakeResponse))
#5 {main}

argilo, 2016-05-29:
@sv1 It looks like you're using Debian's version of CakePHP, which is very outdated. You need to use the version that is installed by `install.sh` (2.8.3).

sv1, 2016-05-29:
I am using the CakePHP that is installed by install.sh, but it is not 2.8.3.

# cake.php --help

Welcome to CakePHP v2.6.7 Console

argilo, 2016-05-29:
OK, you need to update to 2.8.3 then. That version is too old.

sv1, 2016-05-29:
How can I update since pear channel is off???

argilo, 2016-05-29:
Unfortunately you're stuck for now, until the Pear channel comes back up or we figure out another way to install CakePHP.

argilo, 2016-06-01:
The pear channel is back up. Are you able to upgrade now?

sv1, 2016-06-05:
Everything is working as expected. Thank you!!!

please close this issue

closed 2016-06-05 by argilo