urlgrey/hsmm-pi #96: Pear Cake PHP channel website is down, project on hold

cbegg50, 2016-05-29:
Hi, your method of installing Cake PHP relies on the Pear Cake PHP channel and that site is currently down and your install script no longer works.

argilo, 2016-05-29:
Thanks for reporting that. Let's keep an eye on it, and if the service doesn't soon return we can look at changing the way CakePHP is installed.

syelnmh, 2016-05-29:
I noticed this issue yesterday, and it drove me crazy (I missed the errors while running install.sh) . I tried manually installing cake 2.8.3, with no luck. The directory structures don't match up. Test.php was not helpful. From my adventures, I see that cake has 3.x.x versions. I don't expect this project to upgrade, but it should be considered?

argilo, 2016-05-29:
> I don't expect this project to upgrade, but it should be considered?

See issue #85.

argilo, 2016-05-29:
> I missed the errors while running install.sh

I've had this problem too; probably we should add `set -e` to install.sh so it stops anytime an error occurs. I'll open an issue so we remember to do that.

cbegg50, 2016-05-29:
I've modified install.sh and used Composer to install Cakephp V2.8.3 and am currently testing it in my fork.  Just let me know if you want me to keep it.

argilo, 2016-05-29:
Sounds like a good approach, as CakePHP has moved from Pear to Composer in version 3.

argilo, 2016-06-01:
The pear channel is back up now.

syelnmh, 2016-06-01:
The pear/cake setup now installs without error.
I did notice an `E: Unable to locate package ifplugd` during install, not sure if that's related.
Also, cake still doesn't seem to work.

The <pi ip>:8080 page on chrome says `The 192.168.1.75 page isn’t working` (same as when cake wasn't installed properly)
navigating to <pi ip>:8080/hsmm-pi/test.php shows the following:

```

Warning: include(Cake/bootstrap.php): failed to open stream: No such file or directory in /home/pi/hsmm-pi/src/var/www/hsmm-pi/webroot/test.php on line 82
```
```
Warning: include(): Failed opening 'Cake/bootstrap.php' for inclusion (include_path='/home/pi/hsmm-pi/src/var/www/lib:.:/usr/share/php:/usr/share/pear') in /home/pi/hsmm-pi/src/var/www/hsmm-pi/webroot/test.php on line 82
```
```
Fatal error: CakePHP core could not be found. Check the value of CAKE_CORE_INCLUDE_PATH in APP/webroot/index.php. It should point to the directory containing your /cake core directory and your /vendors root directory. in /home/pi/hsmm-pi/src/var/www/hsmm-pi/webroot/test.php on line 91
```

argilo, 2016-06-01:
@syelnmh Are you running the installer on a freshly flashed Debian Jessie image, or something else?

syelnmh, 2016-06-01:
Good call. I've reinstalled the OS, works now. (5th reinstall for hsmm-pi, haha)

argilo, 2016-06-01:
Good to hear.

The installer is built to run on a fresh Raspbian install, and is likely to run into trouble in other situations.

closed 2016-06-05 by argilo