urlgrey/hsmm-pi #103: BBHN Additional Channels

ChrisStory, 2016-06-27:
BBHN now has the ability to use channels/frequencies that are unavailable to consumer WiFi devices. 
Would it be possible to modify this project to use these channels (-1 and -2 in BBHN) ?

argilo, 2016-06-27:
This is something I've been investigating. It's possible, but requires:

* hardware capable of tuning non-standard channels, e.g. Atheros AR9271 chipset
* kernel module patches
* regdb (regulatory database) patches

Lots of useful information is available at http://yo3iiu.ro/blog/?p=1301

I've made the kernel & regdb changes and tuned a TP-Link TL-WN722N to negative channels, but didn't get as far as making HSMM-Pi aware of the negative channels.

My (out-of-date) experiment patching the Raspberry Pi kernel is here: https://github.com/raspberrypi/linux/compare/rpi-3.18.y...argilo:rpi-3.18.y-ath9k-ham

argilo, 2016-06-27:
Of course, recompiling the kernel during HSMM-Pi install would take a ridiculous amount of time, so it would be best to distribute HSMM-Pi as a flashable SD card image if we make a version capable of using negative channels.

mathisono, 2016-06-28:
Your post was great to read... I wasn't completely confident in undertaking
the mod. Ive had the TL-WN722N for a while and love the card... I had some
power consumption problems on the first 2 generations of RPi B. They went
away with the Rpi B with 512mb. My HSMM-pi node is the last thing on ch1.

Do you think you can swing a scripting the kernel and regdb patches. It
should not be part of the the standard build, it could a config flag. Have
the install script ask.

73 Mathison kj6dzb

argilo, 2016-06-28:
@mathisono The blog post isn't mine; it was written by @BogdanDIA. I just followed his instructions.

sv1, 2016-11-28:
Here is a nice "how-to" on how you can compile only one module (ex. the wifi card driver) without compiling the whole kernel

http://askubuntu.com/questions/515407/how-recipe-to-build-only-one-kernel-module

virtualdxs, 2017-03-15:
What's the status on this?

raysuelzer, 2018-11-07:
This is a good idea. I have been poking around a bit in the various drivers source code, the most difficult part of having this integrate with the non part97 AREDN nodes seems like it will be the fact that most of them transmit at 10mhz or 5mhz, and all of the drivers expect either 20ghz or 40ghz bandwidth.