urlgrey/hsmm-pi #104: Why I can not find the DNS node

shaozhai, 2016-06-28:
[image]

shaozhai, 2016-06-28:
My system is armbian

argilo, 2016-06-28:
What is going wrong? Is DNS working?

shaozhai, 2016-06-28:
"dnsmasq: failed to create listening socket for port 53: Address already in use"
[image]

shaozhai, 2016-06-28:
[image]

I can't know which program is occupying the port.

argilo, 2016-06-28:
`sudo netstat -lntup`

shaozhai, 2016-06-28:
[image]

shaozhai, 2016-06-28:
[image]

shaozhai, 2016-06-28:
[image]

Where is my problem?

argilo, 2016-06-28:
It looks like you have both a DNS server and dnsmasq trying to listen on port 53, so you'll have to turn one of them off.

Note that this project only supports Raspbian, not Armbian. So you may have to so some investigation of your own if you want to get this working.

shaozhai, 2016-06-28:
Then how do I uninstall the original DNS service?

argilo, 2016-06-28:
You will have to solve that yourself. I don't know anything about Armbian.

shaozhai, 2016-06-28:
[image]

I don't know that's the default DNS service.

shaozhai, 2016-06-28:
armbian www.armbian.com