urlgrey/hsmm-pi #107: Connect devices to HSMM-Pi mesh network wirelessly

MalvinMiao, 2016-07-21:
So I've established a simple HSMM-Pi mesh network by using only one internal node and one gateway node. Although the connection between mesh nodes is wireless via WiFi, but we can only connect to this mesh network via Ethernet connection? Is that possible I can connect my laptop to the internal node and get Internet access via WiFi connection? Could anybody has the same issue share some detailed introductions about how to do that with me? Thank you guys in advance.

gabdsg, 2016-08-26:
Same question here

ehresman, 2016-08-26:
Olsrd for Windows once worked to connect laptops and the like to HSMM-PI
and BBHN meshes.

However, work stopped on that version of Olsrd years ago, but the Linux
work has continued. Whether it still works with these is going to be
version dependent, I think. I have not personally used Windows Olsrd on a
mesh in at least a couple of years. Recent versions of OpenWRT and other
more current and generic mesh distros were already having compatibility
problems with Windows Olsr when last I used them.

73;
Bob KV4PC