urlgrey/hsmm-pi #110: Mesh node and AP

MalvinMiao, 2016-07-31:
Is that possible that I can make a mesh node operated as an AP at the same time? Since raspberry pi 3's built-in Wi-Fi adapter only support 2.4 GHz band, I don't know how to connect devices with the mesh via Wi-Fi (I know I can connect a router with one internal node and treat the router as an AP). Hope someone can help me figure it out, thank you in advance.

shaozhai, 2016-08-17:
1、hsmm-pi\src\var\www\hsmm-pi\webroot\files\network_interfaces:
ADD：
auto wlan1
allow-hotplug wlan1
iface wlan1 inet static
address 10.20.1.1
netmask 255.255.0.0
#iface wlan0 inet manual
#wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
auto br0
iface br0 inet dhcp
bridge_ports wlan1 wlan0
2、hsmm-pi\src\var\www\curry\webroot\files/dnsmasq:
ADD:
interface=wlan1 
dhcp-option=3,10.20.1.1
dhcp-range=10.20.1.20,10.20.5.200,255.255.0.0,12h
3、apt-get install hostapd bridge-utils
4、/etc/default/hostapd
ADD:
DAEMON_CONF="/etc/hostapd/hostapd.conf"
5、/etc/hostapd/hostapd.conf
ADD:
# Basic configuration 
interface=wlan1
ssid=curry
channel=6
driver=nl80211
ieee80211n=1 
ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40] 
hw_mode=g
wpa=2
wpa_passphrase=12345678
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa_key_mgmt=WPA-PSK 
wpa_pairwise=TKIP
rsn_pairwise=CCMP
----------------------
7、/etc/network/interfaces
ADD:
auto wlan1
allow-hotplug wlan1
iface wlan1 inet static
address 10.20.1.1
netmask 255.255.0.0

auto br0
iface br0 inet dhcp
bridge_ports wlan1 wlan0
9、/etc/dnsmasq/hsmm-pi.conf
interface=wlan1 
dhcp-option=3,10.20.1.1
dhcp-range=10.20.1.20,10.20.5.200,255.255.0.0,1h
10、iptables -t nat -A POSTROUTING -o br0 -j MASQUERADE
11、/etc/sysctl.conf
ADD:
net.ipv4.ip_forward=1
12、/etc/init.d/hostapd start
/etc/init.d/udhcpd start
/etc/init.d/networking restart
hostapd -d /etc/hostapd/hostapd.con

andretansil, 2016-08-19:
@shaozhai can i get your contact ? i cant see any node or make one, thx ! or contact me andretansil@gmail.com

shaozhai, 2016-08-19:
sudo update-rc.d hostapd enable
sudo service hostapd restart
hostapd -d /etc/hostapd/hostapd.conf

andretansil, 2016-08-19:
@shaozhai idk what you mean bro ???

shaozhai, 2016-08-20:
wlan0 is rt3070  wlan1 is rtl8188eu

andretansil, 2016-08-21:
@shaozhai im sorry im OOT, i cant see any mesh in my raspi. i have the same problem as in your page, can you guide me to solve that ? this is your link i mean https://github.com/urlgrey/hsmm-pi/wiki/not-link

andretansil, 2016-08-24:
@shaozhai this script before installation or after installation ?

shaozhai, 2016-08-31:
after

ahmadshkt, 2017-04-29:
@shaozhai I cant find hostapd.conf upon editing it. any suggestions?