urlgrey/hsmm-pi #113: Orange Pi Zero

ch0mik, 2017-01-22:
Hello!

This Project support the Orange Pi Zero (http://www.orangepi.org/orangepizero/) with built-in WiFi ?
	
XR819, IEEE 802.11 b/g/n

Vy 73! de SQ7MRU, Pawel

CarbonC, 2017-02-17:
Thats a really good question, any answers?

tekstrand, 2017-05-01:
I don't see why not. It's just another network adapter.

ilyavy, 2017-06-13:
Are there any confirmed information about working with Orange Pi Zero? I do use them, but after completing the installation and saving settings (for internal mesh node) I, as it is written in manual, rebooted the system, and for some reason the system did not boot at all. The only solution I have found is to reinstall the system...
OS: Armbian

UPD: I have done the same with my friend's Raspberry 3 and got the same result - on screen there are 4 squares (red, yellow, blue and green) and system doesn't want to boot.

Where the problem can be?

tekstrand, 2017-06-13:
@ilyavy can you update your reply with boot logs? https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=23026

jsavage, 2017-06-13:
IIIRC that is a symbol for insufficient current from the power supply. Try
a beefier PSU.

ilyavy, 2017-06-14:
@tekstrand thank you for your quick reply. I updated my post with the logs from /var/log