urlgrey/hsmm-pi #122: hsmm-pi dont work on RPI zero w

Mikkke46, 2017-05-30:
Hello, i want to build a mesh network between two RPI 3 and one RPI zero w.

The RPIs 3 are ok. But i can´t access web application on zero w. The problem maybe is the fact of the zero W don't have ethernet port. Do you have any solution or idea to solve that? What i have to change? 

Regards

2E0PGS, 2017-05-30:
Yes I do. Basically you have to put the PI Zero W modded grub so it will enable Ethernet over USB so it can share internet with the laptop / desktop. 

I may make a video on it since the process is a little tricky. 

The other easier option is to use a PI B by moving the SD card into that for setup then moving it back into the PI Zero W since that shares the same chip set besides WIFI. So you need to add the WIFI dongle to the PI B. Again you need a PI B ofc for that. I haven't tried the latter but I have swapped SD cards between the two a few times before.

Cheers,
2E0PGS

Mikkke46, 2017-05-30:
Do you have any tutorial to follow ? If you could provide the video it will be very nice!  I don´t have any wifi dongle to RPi B...

2E0PGS, 2017-05-30:
Not atm. I plan to do a write up soon and video perhaps.

Mikkke46, 2017-05-30:
Ok thanks! I will try to do as you explain in your 1st reply

natcl, 2017-06-01:
You can setup the pi zero w the following way:

mount the sd card on your computer and modify those files:
in `config.txt` add this at the end: 

```
dtoverlay=dwc2
```

in `cmdline.txt` add this after `rootwait`

```
modules-load=dwc2,g_ether
```

Insert the card in the pi, plug the pi to your laptop using a regular microUSB cable, make sure to use the left-most usb port on the pi.

The pi should be seen as a network interface and you can now ssh to it by it's hostname (`ssh pi@raspberrypi.local`) if you did not change it.  Or you should be able to access the hsmm config page at http://raspberrypi.local:8080