urlgrey/hsmm-pi #128: Updated Shell Script for Raspbian Stretch Lite

bigbydesign, 2017-12-04:
I have updated the install.sh to install properly with Raspbian Stretch Lite. Updated script to use PHP7.0 over PHP5.

[https://rpguy.com/code/install.sh](url)

Nodeswitch, 2018-01-02:
I had to change the permissions on the /var/tmp/oslrd/ to make them writable from the pi user for this to work, otherwise had permission errors thrown at me. 

There might be a better way around doing it, but I simply added:

`sudo chown -R pi:pi /var/tmp/oslrd`

yshbnsl, 2018-01-26:
@bigbydesign your updated script link for php7 is not working.

compuvin, 2018-02-02:
See pull request https://github.com/urlgrey/hsmm-pi/pull/131 for updated script. Changes are similar to those listed on the site listed above but it should install the latest version of PHP regardless of the version.