urlgrey/hsmm-pi #129: Erro in intalation in Raspberry pi 3

pedrofelipest, 2018-01-24:
I followed the installation steps and in sh install.sh gives error.

```
E: Package 'php5' has no candidate for installation
E: 'php5-mcrypt' package has no candidate for installation
E: 'php5-sqlite' package has no candidate for installation
```

compuvin, 2018-02-02:
I made some minor changes to the install script to get mine to work. They are included pull request https://github.com/urlgrey/hsmm-pi/pull/131

Essentially, you need to remove the 5 at the end of all lines with PHP listed and php5-splite should be php-sqlite3

pedrofelipest, 2018-02-02:
I did this, but when I go to configure the network it returns a php error.

compuvin, 2018-02-02:
Is this when you go to submit the data? Did you use the other files in the pull request?

pedrofelipest, 2018-02-02:
This is when I configure the network nodes. I will soon send a screenshot with the error.

schlad21, 2018-12-28:
cd ./hsmmi-pi 
sudo nano install.sh
change the lines with php5 packages to php7 packages
ctrl o
enter
sh install.sh