urlgrey/hsmm-pi #132: Save Network Settings Button: "An Internal Error Has Occured"

Kevin-Ehrichs, 2018-02-13:
In order to get my setup running I have followed the below steps:

1. Used the link in the Readme.md to get the latest Raspian image. 
2. Git clone this repo
3. Changed the following packages in the Install.sh in order to get the install to work
     -php
     -php-mcrypt
     -php-sqlite3

4. After install finished, I was able to load the webpage showing the Node status.

The issue: Whenever I try to change a setting and click the "Save Network Settings"  button, I am always presented with a page that says "An internal Error has occurred."

Even though i get this message, the settings seem to still take effect. 

For example, If I change the node name and "Save Network Settings", I get the error message, but when I refresh the status page, it shows that the node name has indeed changed. 

I am not sure if this is a true issue, which may be leading to not being able to see neighboring nodes.

compuvin, 2018-02-13:
Take a look at my pull request https://github.com/urlgrey/hsmm-pi/pull/131. There is one line of code in the Network Settings page which is causing an issue when saving (has to do with DHCP). I think what might be causing you problems is the update included on that same pull request for a file called rc.local.template that has a few fixes that effect routing.

Kevin-Ehrichs, 2018-02-13:
I was just coming back here to comment and say that i found your pull request! This issue is fixed using your pull request. Thank you.

compuvin, 2018-02-13:
Great! Glad I could help. :) I think you are going to find having a Pi mesh node useful.

Kevin-Ehrichs, 2018-02-13:
Oh wow, so I just made the Pull request #131 changes on my second node and i am now seeing a neighboring node. 

Thank you for all of your help and work to make this work easily on stretch.

Kevin-Ehrichs, 2018-02-13:
I have a few other questions that I dont want to clutter the issues page with. Is there any way I can contact you via email?

compuvin, 2018-02-13:
Sure. I'm still learning myself but happy to help if I can. Replying in a separate comment with my email so I can remove it afterwards.

compuvin, 2018-02-13:
@Kevin-Ehrichs  kb1jdx@gmail.com

closed 2018-02-14 by Kevin-Ehrichs