urlgrey/hsmm-pi #133: RPI3B Internal Wifi Vs. Using External Wifi Dongle.

Kevin-Ehrichs, 2018-02-13:
In the Readme.md file, it states "or built-in WiFi adapter (on Raspberry Pi 3)"

To me this means that If I use the Raspberry Pi 3 B, with internal Wifi (BCM43438 does have ad-hoc capability), then I should not need to purchase an external Wifi dongle. 

I have followed the steps in the readme.md file and have successfully setup 2 nodes. 
Node 1: Wired interface Mode - LAN (Internal Mesh Node)
              WiFi IP: 10.193.65.68
              SSID: Mesh_Test
              Channel: 1

Node 2: Wired interface Mode - WAN( Mesh gateway)
              WiFi IP: 10.39.239.77
              SSID: Mesh_Test
              Channel: 1

**My Issue: Using the internal Wifi Chipset(bcm43438), I cannot see neighboring nodes.** 

I have done plenty of reading and see people with a similar issue. However they end up going out and buying a WiFi dongle. 

Why Do this if the Internal WiFi should be able to do the same? Are the Drivers not developed enough for the internal WiFi? 

**Has any one been able to get this setup working without any external dongles?**

compuvin, 2018-02-13:
I am running the same hardware and do not have an issue with seeing neighboring nodes. Can you verify that the SSID is the same as it is on other nodes? This is on the first Network Settings page under WiFi. The default is for AREDN so if you are using the BBHN software on your other nodes then that needs to be changed on your Pi node to match the others. Tip: click on WiFi scan and copy the SSID from the other network that you want to be a part of and then put that in your WiFi settings area. Save and reboot.

Kevin-Ehrichs, 2018-02-13:
Using the information from pull request #131 I was able to get this to work with internal wifi. I am now seeing neighboring nodes.

closed 2018-02-13 by Kevin-Ehrichs