urlgrey/hsmm-pi #134: Mesh Node using uap0

Kevin-Ehrichs, 2018-02-13:
I have read similar posts about using hsmm-pi as an access point in order to connect other devices wirelessly rather than over the ethernet cable. I have not seen however anyone mention the uap0 virtual interface that the pi3 has. 

I have made a wireless extender from the Pi3 using only the internal wifi chip and no external dongles. Using wlan0 associated with the router, with uap0 as an SSID displaying access point, I was successful and could connect my devices to the internet. 

I would like to try this myself on hsmm-pi as I believe it it feasible to accomplish. Just like the above scenario, the wlan0 interface will be interacting with the mesh network and the uap0 can provide wireless access into the mesh from external devices such as a smartphone. 

Where can I modify the source to do this? Normally, I would configure dnsmasq and hostapd with the /etc/network/interfaces file to get the access point up and running. How does this process differ for HSMM-pi?

suiluj, 2018-04-10:
Did you find a solution? i would like to do the same thing on Pi3.

Kevin-Ehrichs, 2018-05-08:
This is something I would like to eventually work on, Just do not have the time right now.