urlgrey/hsmm-pi #149: Hsmm

jdmaag, 2021-03-29:
Has this project been abandon? 

I can access hsmm pi to configure. Login to the web application on the Pi: http://(wired Ethernet IP of the node):8080/ does not work.
John

compuvin, 2021-04-06:
Unfortunately it seems that way. I personally am still planning on putting some work into this when I have some free time. I haven't installed a new node in a while but have you taken a look at my pull request #131 ? It may still be valid for the latest version of PHP. That sounds like it might be the problem you are running into.

jdmaag, 2021-04-06:
Ok am I supposed to run these 3 scripts

[image: @compuvin] <https://github.com/compuvin>
Update install.sh
<https://github.com/urlgrey/hsmm-pi/pull/131/commits/671bb40aceb4b0228eec07ec8ca6832fa56dffdf>
 …
Verified
671bb40
<https://github.com/urlgrey/hsmm-pi/pull/131/commits/671bb40aceb4b0228eec07ec8ca6832fa56dffdf>
[image: @compuvin] <https://github.com/compuvin>
Update NetworkSettingsController.php
<https://github.com/urlgrey/hsmm-pi/pull/131/commits/dfdd2c6a3c11502da00c7f07e0f7507d126d24b2>
 …
Verified
dfdd2c6
<https://github.com/urlgrey/hsmm-pi/pull/131/commits/dfdd2c6a3c11502da00c7f07e0f7507d126d24b2>
[image: @compuvin] <https://github.com/compuvin>
Update rc.local.template
<https://github.com/urlgrey/hsmm-pi/pull/131/commits/53b3bd1b9c0d91529f074a4f08663de0740f1bc4>
 …
Verified
53b3bd1
<https://github.com/urlgrey/hsmm-pi/pull/131/commits/53b3bd1b9c0d91529f074a4f08663de0740f1bc4>

compuvin, 2021-04-06:
One is a script, the other 2 need to be placed in the appropriate directories. If you don't mind starting over, the whole thing is put together in my forked repo: https://github.com/compuvin/hsmm-pi

jdmaag, 2021-04-09:
OK, i am confused. I would like to start over. I am sort of new to
programming a pi. I have flashed img files before. So what is the
procedure? Do i download the code or just follow the instructions below.
Can you list it out for me please.
John

compuvin, 2021-04-12:
If you're starting with a fresh image, the steps would be as follows:

 sudo apt-get update
 sudo apt-get install -y git
 git clone https://github.com/compuvin/hsmm-pi.git
 cd hsmm-pi
 sh install.sh
 
 After that, you should be able to log onto the web interface at http://(wired Ethernet IP of the node):8080

jdmaag, 2021-04-13:
well i tried it and it still does not let me log in. It replies This site
can not be reached, took to long to reply. I used angry ip to find the ip
address of the raspberry pi, and verified by logging into my router and
checking the ip address. It shows 192.168.1.23. The host name is
raspberrypi.local.  I used http://192.168.1.23:8080
Dont know what else to do.
John

mbroihier, 2021-04-13:
Have you tried connecting from the "mesh" side?

jdmaag, 2021-04-14:
No, how do I do that

mbroihier, 2021-04-14:
One way to do it is to use use a laptop and connect to the wireless network that the PI is talking to/broadcasting on.  I don't happen to recall the default SSID and I believe it has actually changed a time or two, but I suspect you'll see something that looks right.  Mine is AREDN-V2, but I think at one time it was BBHN.  This will be an ad hoc connection and you will need to set your IP address manually.  That address needs to be in the 10.x.x.x range with the mask of 255.0.0.0.  I usually set mine to 10.0.0.1.

If you "ifconfig -a" on your PI, you'll see its IP V4 address on the wlan0 interface.

jdmaag, 2021-04-15:
I tried and found the wlan0 ip but it still will not connect.

Here is a thought.
on the desktop of the pi, on the raspberry pi drop down you need to add a
program folder and put the HSMM in it so it can be accessed that way and
set it up.
Let me know if you do that. As for right now I am at a stand still.
John

mbroihier, 2021-04-15:
If you have run the install.sh script, apache, olsrd, dnsmasq, and gpsd will all start on boot.

I installed the lasted version of urlgrey/hsmm-pi (git commit e5f28b) a few years ago on Raspbian Stretch(4.19.66+ by the way, I always use the lite versions of Raspbian to reduce clutter).  I do recall it being a bit quirky in how the GUI worked, but my setup does not directly connect into my home network so I know part of its quirkiness is routing.

I setup two PI zero W's with this software and have them connected over the mesh network.  A fellow PI hobbyist had told me that he had built a similar configuration and had the PI zeros talking together a half mile apart using only the internal antenna.  I wanted to replicate that and learn more about mesh networks.  So far I haven't replicated his claim.

In my setup, I have one PI zero based at home at a fixed GPS location.  I have another portable one attached to a GPS and it reports the position to base unit.  The base unit logs the position.

Stretch and Buster are different and your problem may be associated with that difference, however, as I said, I did have my own difficulty connecting to the GUI, but connecting to the mesh network eliminated that problem.

Dekymulyana, 2021-04-27:
Hello, sorry before for my bad english.
I have installed 2 raspberry pi with this library HSMM-Pi. 
after install successfull, both of them cant detect neighbor of mesh. did you know what i mistake? 
thanks